start-jsk/rtmros_gazebo #179: Cannot compile hrpsys_gazebo_general

garaemon, 2015-03-14:
I have hrpsys in `ros/hydro_parent`

```
[hrpsys_gazebo_general] /home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/robot.cpp:11:25: fatal error: util/Hrpsys.h: No such file or directory 
[hrpsys_gazebo_general] compilation terminated.                                                                                               
[hrpsys_gazebo_general] /home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/robot.cpp:11:25: fatal error: util/Hrpsys.h: No such file or directory 
[hrpsys_gazebo_general] compilation terminated.                                                                                               
[hrpsys_gazebo_general] /home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/RobotHardware.cpp:13:32: fatal error: util/VectorConvert.h: No such file or directory 
[hrpsys_gazebo_general] compilation terminated.                                                                                               
[hrpsys_gazebo_general] /home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/RobotHardware.cpp:13:32: fatal error: util/VectorConvert.h: No such file or directory 
[hrpsys_gazebo_general] compilation terminated.                                                                                               
[hrpsys_gazebo_general] make[2]: *** [iob/CMakeFiles/RobotHardware_gazebo.dir/home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/robot.cpp.o] Error 1 
[hrpsys_gazebo_general] make[2]: *** Waiting for unfinished jobs....                                                                          
[hrpsys_gazebo_general] make[2]: *** [iob/CMakeFiles/RobotHardwareComp_gazebo.dir/home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/robot.cpp.o] Error 1 
[hrpsys_gazebo_general] make[2]: *** Waiting for unfinished jobs....                                                                          
[hrpsys_gazebo_general] make[2]: *** [iob/CMakeFiles/RobotHardwareComp_gazebo.dir/home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/RobotHardware.cpp.o] Error 1 
[hrpsys_gazebo_general] make[1]: *** [iob/CMakeFiles/RobotHardwareComp_gazebo.dir/all] Error 2                                                
[hrpsys_gazebo_general] make[1]: *** Waiting for unfinished jobs....                                                                          
[hrpsys_gazebo_general] make[2]: *** [iob/CMakeFiles/RobotHardware_gazebo.dir/home/jskuser/ros/hydro_parent/devel/share/hrpsys/src/rtc/RobotHardware/RobotHardware.cpp.o] Error 1 
[hrpsys_gazebo_general] make[1]: *** [iob/CMakeFiles/RobotHardware_gazebo.dir/all] Error 2                                                    
[hrpsys_gazebo_general] make: *** [all] Error 2                                                                                               
```

YoheiKakiuchi, 2015-03-15:
Is it your environment? 
Do you have ``` ~/ros/hydro_parent/devel/include/hrpsys ``` ?

garaemon, 2015-04-21:
Hmm, cannot reproduce it

closed 2015-04-21 by garaemon