start-jsk/rtmros_gazebo #191: STを入れるとおかしな姿勢になる

yoshimalucky, 2015-06-22:
gazeboのsample robotでroseusから

(send *ri* :start-auto-balancer)
(send *ri* :start-st)
とすると、:start-stをした時に、
画像のように左肩が下がります。

:go-posで歩かせると当然コケますが、どこかパラメータがおかしくなっているのでしょうか。

[image]

YoheiKakiuchi, 2015-06-22:
このコミットでセンサが傾いてつけるようになっています。
https://github.com/fkanehiro/openhrp3/commit/c83e28f9cc62c25a13f092f6fccad87448fbeeee

以下のyamlのimuの設定にいい感じでrotationを入れると良いのではないかと思います。
https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/config/SampleRobot.yaml#L122

yoshimalucky, 2015-06-22:
imu_sensor0: {ros_name: 'sample_imu_sensor', link_name: 'WAIST_LINK0', frame_id: 'WAIST_LINK0', translation: [0, 0, 0], rotation: [1, 0, 0, 0]}
のように追記して、rotationの値をいい感じに変えてやるということだと思うのですが、
そもそもopenhrp3の方でこのような変更が加えられたのはどうしてでしょうか。
openhrp3の方でこのままセンサが傾いてついた状態は維持されて、
こっちでrotationを入れてpullreqすればいいですか？

eisoku9618, 2015-06-22:
> そもそもopenhrp3の方でこのような変更が加えられたのはどうしてでしょうか。

kfのデバッグ用に僕が加えました．

hrpsys-simulatorでは動いていたので大丈夫だったと思っていましたが…

eisoku9618, 2015-06-22:
歩かないのはstep timeを小さくすると歩くと思います。

https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/worlds/empty.world#L20

また、ちょっとあれですが、``(send *ri* :calibrate-inertia-sensor)``したら真っ直ぐになります。

k-okada, 2015-06-22:
というか，そもそも，imu_sensors_configが有るのはなぜだろう．
センサの情報がcolladaにはいっていて，それがgazeboに取り込まれていれば，
この情報は必要ないと思うんだけど．．．

YoheiKakiuchi, 2015-06-22:
> それがgazeboに取り込まれていれば，

これがむずかしそうで，実装されていないです．
ですが，yamlに書いてある情報でimuが一番現実的にできそうではあります．

k-okada, 2015-06-22:
ふーむ．pr2みるとurdfから spawn_modelしていて，imuがみえているから，簡単そうだけどなぁ．

yoshimalucky, 2015-06-23:
(send *ri* :calibrate-inertia-sensor) したらまっすぐになりました。
ありがとうございます。とりあえずはこれで動かそうと思います。

k-okada, 2015-06-24:
本来の趣旨とは違う方向で定着してしまうと良くないのでちゃんと治したいところです．

snozawa, 2015-06-24:
これは、collada=>urdfに変換したときにセンサ情報が残らないということでしょうか。
それともurdfにセンサ情報をかいてもgazeboで読み込まれないということでしょうか。

k-okada, 2015-06-24:
ちゃんとはわかっていませんが，
SampleRobot.yamlにちゃんと書いていないから，手で書いてコミットするは最低限として，
このファイルが自動生成なら部分に位置姿勢がちゃんと入るように直すのがいいのと，
urdfにgazeboのimuがはいっていないなら，それをいれるのがいいでしょう．
という感じです．

snozawa, 2015-06-24:
pr2がimuをよんでるのも、とくに（PR２用の）gazeboプラグインがあるのでなくて、
gazebo正規の機能として読み込んでimu値を出してるかんじでしょうか。
であれば特殊なことをrtmros_gazeboでしなくてもでてきそうですね。。。

k-okada, 2015-06-24:
https://github.com/PR2/pr2_common/blob/indigo-devel/pr2_description/urdf/sensors/microstrain_3dmgx2_imu.urdf.xacro
から
https://github.com/PR2/pr2_common/blob/indigo-devel/pr2_description/urdf/sensors/microstrain_3dmgx2_imu.gazebo.xacro
を読んで
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_imu.cpp
を使っているように見えます．

YoheiKakiuchi, 2015-06-24:
> これは、collada=>urdfに変換したときにセンサ情報が残らないということでしょうか。
> それともurdfにセンサ情報をかいてもgazeboで読み込まれないということでしょうか。
両方ですね。
前者は、urdfのテキストにはgazeboタグが書けるけど、ROSの側では読み込まないので、collada_to_urdfするときに、ROSで一度読んだ後にcolladaをもう一度パースしてurdfに追加する必要があります。

後者は、現状コードがないです。
PR2の方式を使うとROSのトピックとしてimuが出てしまいます。
現状では、IOBPlugin(gazebo) <-> iob(hrpsys)のみが通信することでhrpsysとgazeboの同期をとっています。
ですが、複数個imuがなければそれほど問題なくIOBPluginで書けると思います。

YoheiKakiuchi, 2015-06-24:
https://github.com/start-jsk/rtmros_tutorials/pull/332 で修正しています。

ちょっと間違ったことを書きました。
urdfにxacroでセンサ情報を書いています。iobで読むためにsensor_id(センサの順番)だけimu_sensors_configに書く必要があります。

snozawa, 2015-06-24:
> 前者は、urdfのテキストにはgazeboタグが書けるけど、ROSの側では読み込まないので、collada_to_urdfするときに、ROSで一度読んだ後にcolladaをもう一度パースしてurdfに追加する必要があります。

前者の問題は、`collada_to_urdf`がセンサを読み込まないので、その生成物にもセンサが含まれない、そのためあとで別途追加が必要ということでしょうか。

> 後者は、現状コードがないです。
> PR2の方式を使うとROSのトピックとしてimuが出てしまいます。
> 現状では、IOBPlugin(gazebo) <-> iob(hrpsys)のみが通信することでhrpsysとgazeboの同期をとっています。
> ですが、複数個imuがなければそれほど問題なくIOBPluginで書けると思います。

`/imu`がトピックとしてでていると何が問題でしょうか。
remapすれば良い？

> start-jsk/rtmros_tutorials#332 で修正しています。

こちらも手書きでかかずに、正規のmodel conversionの流れでできるものでしょうか。
また、idはいれられそうじゃないでしょうか？
http://wiki.ros.org/urdf/XML/sensor/proposals
http://sdformat.org/spec?ver=1.5&elem=sensor

あとこれに関連して、もしかしたらcolladaを経由する方法をやめてurdfを経由するように色々書き換えたら
それで良い気もしてきています。