start-jsk/rtmros_gazebo #195: gazebo用のurdfモデルが実機モデルと違う(カメラ、endcoords)

garaemon, 2015-06-22:
カメラやendcoordsが含まれていない、など。
実機モデル用のurdfにgazeboタグを入れちゃうのでどうでしょうか?

mmurooka, 2015-06-22:
collada_urdf_jsk_patch を使わないと，gazeboに正しくモデルが表示されなくて(https://github.com/start-jsk/rtmros_gazebo/issues/48 , https://github.com/start-jsk/rtmros_gazebo/issues/47 )，

hrpsys_ros_bridge_tutorialsでは，collada_urdfを使っていて（ https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/cmake/compile_robot_model.cmake#L320 ），
hrpsys_gazebo_generalでは，collada_urdf_jsk_patchを使っています（ https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/cmake/compile_robot_model_for_gazebo.cmake#L29 ）．

一番良さそうな解決策は，
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/cmake/compile_robot_model.cmake
の compile_openhrp_model_for_closed_robots, compile_rbrain_model_for_closed_robots などの引数で，
collada_urdfのパッケージ名を与えられるようにしてデフォルトはcollada_urdf, hrpsys_gazebo_tutorialsの下ではcollada_urdf_jsk_patchとして，これを呼ぶ，
でしょうか．

k-okada, 2015-06-23:
変なifdefいれると後で大変になってきてしまうので，全部jsk_patchでよくて，indigoはリリースされていないならリリースする方向でいい気がしますが，
そもそも僕の理解が正しければ，collada_urdf_jsk_patchはcollada_urdfをassimp-develに対してリンクしたものでこれは形状モデルの変換だけにつかっているので
おかしければgazebo/rviz関係なくurdfが変になっているという理解はただしいだろうか？
もしgazeboだけ正しいとするとgazeboタグがおかしいので、それはcollada_urdfの問題になるきがする

ところこれはhrpのgazeboはurdfからうごいているんだったね。とするとimu_sensors_configはますます必要ないと思うけど。ちがうかな.

snozawa, 2015-06-23:
> カメラやendcoordsが含まれていない、など。
> 実機モデル用のurdfにgazeboタグを入れちゃうのでどうでしょうか?

実機とsimulationで同じurdfになれば良いですね。

mmurooka, 2015-06-23:
> もしgazeboだけ正しいとするとgazeboタグがおかしいので、

gazeboだけおかしいのですが，問題があるのはgazeboタグではなさそうです．
試しにhrp2jskntで
hrpsys_ros_bridge_tutorialsの下のurdf
（`$(find hrpsys_ros_bridge_tutorials)/models/HRP2JSKNT.urdf`, `$(find hrpsys_ros_bridge_tutorials)/models/HRP2JSKNT_WH.urdf`, `$(find hrpsys_ros_bridge_tutorials)/models/HRP2JSKNT_WH_SENSOR.urdf` ）
などをrobot_descriptionにしてgazeboにspawn（ https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/launch/gazebo_robot_no_controllers.launch#L61-L63 ）すると，
https://github.com/start-jsk/rtmros_gazebo/issues/47#issue-32613940
のエラーがでて表示されませんでした．

hrpsys_gazebo_tutorials と hrpsys_ros_bridge_tutorials の下のurdf（collada_urdfとcollada_urdf_jsk_patchで生成されたurdf）の間のdiffは，
主にdaeのmeshだけだったので，
> collada_urdf_jsk_patchはcollada_urdfをassimp-develに対してリンクしたものでこれは形状モデルの変換だけにつかっているので

のpatchがgazeboには必要になっているようです．

k-okada, 2015-06-23:
> #47 (comment)

Assimpのどのバージョンのどの変更がきいているかわかるかな

どこかの時点でjsk-patchはなくしたいので、そに目安に情報がほしい

mmurooka, 2015-06-26:
> Assimpのどのバージョンのどの変更がきいているかわかるかな

@YoheiKakiuchi さんに聞いて，
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/jsk_ros_patch/collada_urdf_jsk_patch/use_assimp_devel.patch
のパッチだそうです．
ubuntuのassimpのバージョンの問題なので，
collada_urdf_jsk_patchに反映してもらって解決，とはできないようです．

k-okada, 2015-06-26:
そのUbuntuのassimpバージョンがどれだったら何ができて何ができないかな知りたいです

とりあえずhttps://launchpad.net/ubuntu/+source/assimp
でみえている範囲だとassimpは3.0で
assimp_develは3.1なので当面は必要なのかな

k-okada, 2015-06-26:
3.0と3.1の差分はなんだろう

YoheiKakiuchi, 2015-06-26:
12.04ではassimpのバージョンが低くてexport-apiが使えないのだと思われます。
https://github.com/ros/robot_model/blob/indigo-devel/collada_urdf/CMakeLists.txt#L27-L29

3.0になればassimpで作られたファイルになるので、14.04で使えるか確認すればいいかと思います。
現状のcollada_to_urdfでは以下でダンプされたメッシュが使われます。
https://github.com/ros/robot_model/blob/indigo-devel/collada_parser/src/collada_parser.cpp#L1235-L1337

garaemon, 2015-08-26:
実機とシミュレーション同じにしました。

抜けがあるのでカメラ周りが動かなくなっている予感もします