start-jsk/rtmros_gazebo #205: SampleRobot.confが/optのパスを書いている

garaemon, 2015-08-20:
ソースからインストールしていると動かない
http://wiki.ros.org/rtmros_common/Tutorials/HrpsysGazebo

https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/config/SampleRobot.conf

k-okada, 2015-08-20:
https://github.com/fkanehiro/hrpsys-base/pull/596
https://github.com/fkanehiro/openhrp3/issues/55#issuecomment-94127875

たぶんだけど，PROJECT_DIRとかが使えるんじゃないかと思います．

garaemon, 2015-08-20:
configureするよりPROJECT_DIR使ったほうがいいですかね?

https://github.com/garaemon/rtmros_gazebo/blob/samplerobot-configure/hrpsys_gazebo_general/config/SampleRobot.conf.in
https://github.com/garaemon/rtmros_gazebo/blob/samplerobot-configure/hrpsys_gazebo_general/CMakeLists.txt#L90

k-okada, 2015-08-20:
configureはやばくない？

garaemon, 2015-08-20:
コンパイル時にOpenHRP3のパスを解決してコピーしているだけなので、特に問題ない気がします

◉  ryohei

k-okada, 2015-08-20:
勿論わかっている人はいいんだろうけど，この手のものは，自動生成ファイルがドンドン出来て何がなんだかわからんくなって，結局生成物のファイルを変更してコミットする，という方向にしか行かない気がする．せっかくPROJECT_DIRとかあるんだから，うまく使える（使えるようにする）のがいいと思います．

garaemon, 2015-08-20:
confで`model: $(PROJECT_DIR)/../model/sample1.wrl`
にしてみましたが、
```
loading $(PROJECT_DIR)/../model/sample1.wrl
model/sample1.wrl cannot be found.
Retrying to load the file as a standard VRML file
model/sample1.wrl cannot be found.
loading failed.
model/sample1.wrl cannot be found.
ModelLoaderException : model/sample1.wrl cannot be found.
[hes] failed to load model[$(PROJECT_DIR)/../model/sample1.wrl]
```
といわれますね.
xmlか何かに書かないといけないですか

garaemon, 2015-08-20:
$PROJECT_DIRが効くのはhrpsys_config.pyとProject.cppだけでしょうか.

k-okada, 2015-08-20:
なんと，そうでした．ということで，https://github.com/fkanehiro/openhrp3/pull/68

garaemon, 2015-08-21:
solved by https://github.com/fkanehiro/openhrp3/pull/68 and https://github.com/start-jsk/rtmros_gazebo/pull/211

closed 2015-08-21 by garaemon