start-jsk/rtmros_gazebo #222: gazebo上でHRP2のグリッパで物体を把持する

mmurooka, 2015-12-09:
before
[image]
滑っていく．

after https://github.com/start-jsk/rtmros_tutorials/pull/449
[image]
滑っていかなくなったが，めり込んでいく．

mmurooka, 2015-12-09:
https://github.com/PR2/pr2_common/blob/indigo-devel/pr2_description/urdf/gripper_v0/gripper.transmission.xacro#L18-L46
でグリッパ専用のtransmisionが使われています．
PR2のグリッパの機構に特化しているように見えるので，
そのままHRP2のグリッパに使い回しはきかなさそうです．

HRP2についてグリッパが物体にめり込まないようにするための解決案は，

１．
https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/src/IOBPlugin.cpp
でやっている関節角の制御に関して，
グリッパ関節では電流制御・トルク制御みたいなことをやる．

２．
HRP2のグリッパ専用のtransmisionを作る．
（transmisionが何をしているのかよくわからないので，
これに意味があるかはわかっていないです．）

でしょうか．

k-okada, 2015-12-09:
このハンドはオリジナルなので滑っていくのはOKにみえる．
JSKでつかっていたのは，ちょっと長くしたものなので，形状が違う．
CADかEUSで適当にモデルつくって試してみたらいいと思う．

k-okada, 2015-12-09:
滑るバージョンでいいので，ContactとForceの表示と視野画像を含めたビデオができると嬉しいです．

mmurooka, 2015-12-09:
> 滑るバージョンでいいので，ContactとForceの表示と視野画像を含めたビデオができると嬉しいです．

https://drive.google.com/a/jsk.imi.i.u-tokyo.ac.jp/file/d/0B5qanGnXorOuVG5wM290WEVDZEU/view?usp=sharing

k-okada, 2015-12-09:
良いのだけど，カメラのcrop面が遠すぎないかな．あとcontactの線はでているのかな．

mmurooka, 2015-12-09:
> 良いのだけど，カメラのcrop面が遠すぎないかな．あとcontactの線はでているのかな．

クロップを近くして，contactの線を出すようにしました．
https://drive.google.com/a/jsk.imi.i.u-tokyo.ac.jp/file/d/0B5qanGnXorOuMkhUUlNyNDNkRDQ/view?usp=sharing

Rvizに手先力センサ情報も表示してみましたが，
コップに触れても目で見て分かる変化はなく振動しているだけだったようなので非表示にしました．

クロップについては，
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_gazebo_tutorials/robot_models/HRP2JSK/HRP2JSK.urdf.xacro#L81-L84
にあって，
今回の動画では，0.5を0.2にしました．
ポイントクラウドとカメラで共通の値を使う仕様になってしまっています．