start-jsk/rtmros_gazebo #226: Build failed in Jenkins: Idoc__rtmros_gazebo__ubuntu_trusty_amd64

mmurooka, 2016-03-20:
http://build.ros.org/job/Idoc__rtmros_gazebo__ubuntu_trusty_amd64/1/

```
05:19:26 hrpsys_gazebo_atlas has BUILD message dependencies:
05:19:26   - roseus
05:19:26   - hrpsys_ros_bridge
05:19:26   - tf
05:19:26   - laser_assembler
05:19:26   - image_view2
05:19:26   - jsk_pcl_ros
05:19:26   - laser_assembler
05:19:26 hrpsys_gazebo_atlas has BUILD dependencies:
05:19:26   - roseus
05:19:26   - pr2eus
05:19:26   - hrpsys_ros_bridge
05:19:26   - pr2_controller_manager
05:19:26   - collada_urdf_jsk_patch
05:19:26   - tf
05:19:26   - laser_assembler
05:19:26   - image_view2
05:19:26   - jsk_pcl_ros
05:19:26   - collada_urdf_jsk_patch
05:19:26   - resized_image_transport
05:19:26   - hrpsys_gazebo_general
05:19:26   - laser_assembler
05:19:26   - message_generation
05:19:27 CMake Error at /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:260 (message):
05:19:27   Messages depends on unknown pkg: jsk_footstep_msgs (Missing
05:19:27   find_package(jsk_footstep_msgs?))
05:19:27 Call Stack (most recent call first):
05:19:27   CMakeLists.txt:91 (generate_messages)
05:19:27 
05:19:27 
05:19:27 -- Configuring incomplete, errors occurred!
```

jsk_footstep_msgs is depended by hrpsys_gazebo_atals.
dependency order is hrpsys_gazebo_atlas -> jsk_pcl_ros -> jsk_footstep_msgs.

I generated 14.04/indigo environment on docker, and build rtmros_gazebo source; other repository is installed via rosdep. catkin build is passed, and I cannot reproduce the error.

Can I get hints to solve or reproduce error?

k-okada, 2016-03-22:
please check if you have same jsk_pcl_ros version .

mmurooka, 2016-03-31:
deb build is passed by updating release source.

closed 2016-03-31 by mmurooka