start-jsk/rtmros_gazebo #233: RobotHardwareComp is not installed as debian

furushchev, 2017-01-10:
In https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/iob/CMakeLists.txt#L26 and https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/iob/CMakeLists.txt#L39, `RobotHardwareComp` is specified to be installed, but no such library is actually installed.

```bash
$ dpkg -L ros-indigo-hrpsys-gazebo-general | grep Robot
/opt/ros/indigo/share/hrpsys_gazebo_general/config/SampleRobot.yaml
/opt/ros/indigo/share/hrpsys_gazebo_general/config/SampleRobot.conf
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/SampleRobot_gazebo.urdf
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/SampleRobot_additional_urdf_setting.sh
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/model.config
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/SampleRobot.urdf.xacro
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LLEG_LINK6_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RARM_LINK7_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RLEG_LINK4_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RLEG_LINK3_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LLEG_LINK5_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RLEG_LINK1_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LARM_LINK2_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RARM_LINK1_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LLEG_LINK1_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/WAIST_LINK0_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/WAIST_LINK2_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LLEG_LINK3_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RARM_LINK4_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LARM_LINK1_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RLEG_LINK5_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RARM_LINK2_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LLEG_LINK4_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RLEG_LINK6_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LARM_LINK3_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RARM_LINK6_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/WAIST_LINK1_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LARM_LINK7_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LARM_LINK4_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/WAIST_LINK3_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LARM_LINK6_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RARM_LINK3_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LARM_LINK5_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RARM_LINK5_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/RLEG_LINK2_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/meshes/LLEG_LINK2_mesh.dae
/opt/ros/indigo/share/hrpsys_gazebo_general/robot_models/SampleRobot/SampleRobot.urdf
```

mmurooka, 2017-01-10:
Thank you! If anyone knows how to fix this, please tell us.

furushchev, 2017-01-11:
As far as we can see in previous issue (https://github.com/start-jsk/rtmros_gazebo/issues/35), we could solve this problem if we have this package from source.
After compile, we have `lib` folder in `hrpsys_gazebo_general` package and `RobotHardware.so` in it, but this is not installed.
The problem is `hrpsys` package also provides `RobotHardware.so` (the same name) for real robot and we expect dependee application overloads to use `RobotHardware.so` of this package (not of `hrpsys`).

furushchev, 2017-01-11:
**MEMO** `RobotHardware.so` is searched according with this line https://github.com/start-jsk/rtmros_gazebo/blob/master/hrpsys_gazebo_general/launch/robot_hrpsys_bringup.launch#L46

PR #234 (2017-01-11): [hrpsys_gazebo_general][iob/CMakeLists.txt] fix: remove build type spec from install
This PR will fix #233 

When releasing as deb `CMAKE_BUILD_TYPE` is set to None (not Release, Debug  nor RelWithDebInfo)
closed 2017-01-11 by k-okada